njh/EtherCard #291: webClient example not working

abdullahseba, 2017-11-18:
Hi I tried the default webClient google example and all I get is:
```
<<< REQ >>>
HTTP/1.0 404 Not Found
Content-Type: text/html; charset=UTF-8
Referrer-Policy: no-referrer
Content-Length: 1568
Date: Sat, 18 Nov 2017 20:32:27 GMT

<!DOCTYPE html>
<html lang=en>
  <meta charset=utf-8>
  <meta name=viewport content="initial-scale=1, minimum-scale=1, width=device-width">
  <ti...
```

I also tried it with my own server for what I actually need to use it for  and I get a 400 error

```
<<< REQ >>>
HTTP/1.1 400 Bad Request
Content-Type: text/html; charset=us-ascii
Server: Microsoft-HTTPAPI/2.0
Date: Sat, 18 Nov 2017 20:44:08 GMT
Connection: close
Content-Length: 311

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01//EN""http://www.w3.org/TR/html4/strict.dtd">
<HTML><HEAD><TITLE>Bad Request</TITLE>
<META HTTP-EQUIV="Content-Type" Content="text/html; charset=us-ascii"></HEAD>
<BODY><h2>Bad Request</h2>
<hr><p>HTTP Error 400. The request is badly formed.</p>
</BODY></HTML>
```
I've also other tried `ether.browseUrl(PSTR("/index.php"), "", website, my_callback);` with the same result. Is this a bug? How to I make a working get request and print the return to serial? 
Thanks

mucahitkurtlar, 2020-04-05:
Did you find any solution?

artemdudkin, 2020-04-06:
Looks like it works and something wrong with request (as server returns 400). You can log `buffer` at `ENC28J60::packetSend` to see what actually happens (when I did this I realized that "Host:" is empty because ether.browseUrl demands third arg (i.e. website) to be PROGMEM variable).