njh/EtherCard #294: httpServerReply closes TCP connection

xinthose, 2017-11-29:
Hello,
    I am replying to a ping command from boost ASIO with the following code:

```
ether.persistTcpConnection(true);    // done in setup function

const char tcpReply[] PROGMEM = "Received\n";
//Copy Reply To TCP Buffer
memcpy_P(ether.tcpOffset(), tcpReply, sizeof tcpReply);
//Sends reply "Received" to the sender to acknowlege receipt of the packet
ether.httpServerReply(sizeof tcpReply - 1);
```

I get the first response, but then get stuck in an infinite EOF loop with `async_read_until`.  Is the connection being closed?  How can I keep it open?  

I read until I see '\n' then run `async_read_until` to give the io_service work.

Thank you.

njh, 2018-07-28:
Sorry, several months later and it is looking unlikely anyone is able to help with your query.

Closing the issue, but feel free to re-open if you find a solution or someone is able to help.

closed 2018-07-28 by njh

xinthose, 2018-07-30:
OK.  Arduino's official Ethernet 2 library worked for me: https://www.arduino.cc/en/Reference/Ethernet

njh, 2018-07-30:
I am glad you got something working.

EtherCard has a very minimal implementation of TCP/IP that runs on the very resource constrained Arduino micro controller. Then ENC28J60 just handles the Ethernet communication. It is not unexpected that some uses of it will fail.

With the Arduino Ethernet library, the W5100/W5500 chips runs a far more complete implementation of TCP/IP and the library acts as an interface to the chips API.