njh/EtherCard #295: How to connect to Arduino Leonardo?

braian87b, 2017-12-02:
Hi, I have an Arduino Leonardo and I want to know how to connect the board to the Leonardo.
I google it about but there is about 5 or 6 different connection schemes, I have tried the same as for UNO and another one, I do not want to keep trying to avoid damage the board...
Can you point me out how I need to connect and what I need to do to be able to get it working? 
Thanks in advance.

njh, 2018-07-27:
Hi,

I am not sure what you mean by 'connect'?
Is it a ENC28J60 shield?
Or a separate board that you are trying to connect via jumper cables?

Same connections as Uno should work fine.. but maybe provide a link to what you are trying to do?

nick.

njh, 2018-08-28:
No response, so closing issue.

closed 2018-08-28 by njh