njh/EtherCard #296: How often I need call ether.packetLoop(ether.packetReceive()) ?

Dravon5X, 2017-12-06:
So as I found is necessary to call ether.packetLoop(ether.packetReceive()) otherwise it won't work.

But how often I need call it? Now I have it in loop() and with millis and timer I'm waiting when to send data. But before sending data, I'm calling temp reading function, this function can take 5-10 seconds (reading of one sensor 0.7 seconds, if sensors are disconnected it wait 5 seconds and recheck again).

After returning from that 5 seconds function how often I need call ether.packetLoop(ether.packetReceive()) until I can send data?

I made some experiments, but it's not clear, sometimes it work sometimes not.

thanks

njh, 2017-12-06:
You need to call it often enough that the packet buffer in the ENC28J60 ethernet controller doesn't overflow. If it overflows then packets will be lost. The buffer in the ENC28J60 is 8Kbytes - enough for around 8 'full size' Ethernet frames.

I call it once for every time round in the main Arduino `loop()` and avoid doing anything that would cause any delays, such as sleeps.

Dravon5X, 2017-12-06:
the reading of temperature sensor take 0.7 sec and I have two of them, but maybe i can wait after reading 5 seconds in loop where I will call packetLoop and send data to my server after.

Anyway can you add some text about this and put it in examples? Just lke you need avoid delay(), use micros() instead and call packetLoop every loop.

thanks

closed 2018-07-28 by njh