njh/EtherCard #298: change another pin

beryindo, 2017-12-08:
//   SCK - Pin 13
//   SO  - Pin 12
//   SI  - Pin 11
//   CS  - Pin  8

can change pin 8 & pin 11 UNO ? because pin 8 & pin 11 has been use for lcd tft.

njh, 2017-12-08:
Does your LCD TFT use SPI?

njh, 2017-12-08:
SPI allows you to share MISO, MOSI and SCK between multiple devices.
Only CS has to be unique per device.

The CS pin can be chosen when calling `begin()`:

```
    /**   @brief  Initialise the network interface
    *     @param  size Size of data buffer
    *     @param  macaddr Hardware address to assign to the network interface (6 bytes)
    *     @param  csPin Arduino pin number connected to chip select. Default = 8
    *     @return <i>uint8_t</i> Firmware version or zero on failure.
    */
    static uint8_t begin (const uint16_t size, const uint8_t* macaddr,
                          uint8_t csPin = SS);
```

https://jeelabs.org/pub/docs/ethercard/classEtherCard.html#ab195820843fba5f3b465eeaaf7fc9daa

beryindo, 2017-12-09:
thank you, I use pin 8 to pin 10.
how to browseUrl used button ? I try not work

njh, 2018-07-27:
Sorry, I am not sure what "how to browseUrl used button ?" means.

Please open a new issue if you are still having trouble.

closed 2018-07-27 by njh