njh/EtherCard #300: Telnet Server with EtherCard

Noobio, 2018-01-22:
Hello! Thank You for great arduino library!
Help me please to write a simple version of Telnet server for Arduino Nano.
Is it possible with EtherCard?
I don't want to use UIPEthernet library, becouse It's too heavy for Arduino Nano and I can't compile all my sketch.
I need to control two pins of Arduino.
Help me please! Thank You!

njh, 2018-01-22:
I do not believe it is currently possible to create a Telnet server using EtherCard. One of the reasons is that long running TCP connections require more RAM to track packet resends.

Could you use an HTTP server or UDP instead?

Noobio, 2018-01-22:
I already use HTTP in my project.
May be it is possible to make a small prototype with a short session time?

njh, 2018-07-27:
By long-running I mean a TCP connection that isn't as simple as:

1. Request
2. Response
3. Close Socket

If you want to use Telnet, another option is to use a uIP based library such as UIPEthernet:
https://github.com/UIPEthernet/UIPEthernet

I don't think it is realistic to create a Telnet Server with EtherCard, so I am closing this ticket.

closed 2018-07-27 by njh