njh/EtherCard #301: Get SOA record from DNS

andli, 2018-01-25:
Hello! I have tried reading through the source code for a way to get the SOA record, but it seems like all DNS reply data is thrown away except the IP address.

My real target is to find out which domain the network I plug into presents itself as.

Would it be easy for you to add something that corresponds to `nslookup 192.168.1.1` on Windows?

njh, 2018-01-29:
The DNS code is here:
https://github.com/jcw/ethercard/blob/master/dns.cpp

To keep the code size small it is hard-coded to doing 'A' record lookups.
While I don't think it would be easy to modify the code to support generic lookups, it should be possible for you to generate the requests and parse the responses yourself by sending the UDP packets.

Alternatively have you considered getting the domain name from DHCP?

Maybe Option 15 or 119?
https://www.iana.org/assignments/bootp-dhcp-parameters/bootp-dhcp-parameters.xhtml

njh, 2018-07-28:
I think it is unlikely that support for this will get implemented any time soon.

Closing this issue but would welcome a Pull Request that added support for this.

closed 2018-07-28 by njh