njh/EtherCard #303: Mega & enc28j60

Comandante77, 2018-02-16:
Cant connect enc28j60 module to my Mega. 

After 
`if (ether.begin(sizeof Ethernet::buffer, mymac, 53) == 0)
  {
    Serial.println(P("Failed to access Ethernet controller"));
  }
  else
  {
     Serial.println(P("Ethernet is fine and up"));
  `}``

I'm getting "Failed to access Ethernet controller".

DHCP Failed also.

Leds on enc28j60 are working. Link is on and Ack is blinking.

Tried to use external power supply for 3.3V, the same result.
[image]
[image]

Ruttmann, 2018-03-14:
Are you sure you're connection all the necessary pins? I also using ENC28J60 with MEGA and only the pins CLKOUT and WOL aren't in use.  I can initialize and connect to the Ethernet controller but I'm still wondering why DHCP doesn't works...

njh, 2018-07-28:
Closing old tickets, please re-open if you still interested in this issue.

closed 2018-07-28 by njh

thangamuthun, 2018-07-29:
here i am also facing same problem for past two weeks.
When i am uploading backsoon program , Failed to configure Ethernet is displayed in Serial monitor.
help me!!

njh, 2018-09-06:
This issue is closed.

Please can you raise a new issue explaining your problem and what you are tried.