njh/EtherCard #309: send udp to specific ip address instead of broadcast

Robbert96, 2018-05-01:
I cant seem to send a udp message to a specific ip address. broadcasting at 255.255.255.255 does work. Ive seen other issues address this problem, but have not seen a resolution. I dont see the packet appear on WireShark, which suggests it is not a firewall problem right?  any help is greatly appreciated. this is my arduino code:

`
// Demonstrates usage of the new udpServer feature.
//You can register the same function to multiple ports, and multiple functions to the same port.
//
// 2013-4-7 Brian Lee <cybexsoft@hotmail.com>

#include <EtherCard.h>
#include <IPAddress.h>

#define STATIC 1  // set to 1 to disable DHCP (adjust myip/gwip values below)

#if STATIC
// ethernet interface ip address
static byte myip[] = { 192,168,178,180 };
// gateway ip address
static byte gwip[] = { 192,168,178,1 };
static byte dnsip[] = { 84,116,46,21 };

unsigned int myPort = 8888;
unsigned int destPort = 9000;
static byte destIP[] = { 192,168,178,27 }; //255,255,255,255

#endif

// ethernet mac address - must be unique on your network
static byte mymac[] = { 0x70,0x69,0x69,0x2D,0x30,0x31 };

byte Ethernet::buffer[500]; // tcp/ip send and receive buffer
char data;
char msg[] = {"Hello World"};

//callback that prints received packets to the serial port
void udpSerialPrint(uint16_t port, byte ip[4], uint16_t server, char *data, uint16_t len){
  IPAddress src(ip[0], ip[1], ip[2], ip[3]);
  Serial.println("Received a message!");
  Serial.print("From: ");
  Serial.println(src);
  Serial.print("On port: ");
  Serial.println(port);
  Serial.print("UDP Server: ");
  Serial.println(server);
  Serial.print("Length of message: ");
  Serial.println(len);
  Serial.print("Message: ");
  Serial.println(data);  
  ether.sendUdp(msg, sizeof msg, myPort, destIP, destPort);

} 

void setup(){
  Serial.begin(57600);
  Serial.println("\n[backSoon]");

  if (ether.begin(sizeof Ethernet::buffer, mymac,53) == 0)
    Serial.println( "Failed to access Ethernet controller");
#if STATIC
  ether.staticSetup(myip, gwip, dnsip);
#else
  if (!ether.dhcpSetup())
    Serial.println("DHCP failed");
#endif

  ether.printIp("IP:  ", ether.myip);
  ether.printIp("GW:  ", ether.gwip);
  ether.printIp("DNS: ", ether.dnsip);

  //register udpSerialPrint() to port 1337
  ether.udpServerListenOnPort(&udpSerialPrint, myPort);

//  //register udpSerialPrint() to port 42.
//  ether.udpServerListenOnPort(&udpSerialPrint, 42);
}

void loop(){
  //this must be called for ethercard functions to work.
  ether.packetLoop(ether.packetReceive());
}
`

farmerbriantee, 2018-07-26:
Have the same problem. When you look at the sender endpoint in receiving application all is good, the port number and IP are correct. Set destination to the receiving apps IP, wont send. While not the end of the world, I have several devices doing this and it certainly adds a lot of unnecessary traffic and processing for all the modules.

metinkun, 2018-09-20:
i had the same problem. after setting mask problem solved.

static byte myip[] = { 192,168,178,180 };
static byte gwip[] = { 192,168,178,1 };
static byte dnsip[] = { 84,116,46,21 };
static byte mask[] = { 255,255,255,0 };
.
.
.

ether.staticSetup(myip, gwip, dnsip , mask);

njh, 2018-09-20:
Thanks @metinkun. Good spot!

Yes, without the subnet mask, then EtherCard won't know if the IP address is on the local subnet or not.

@farmerbriantee would be great to know if it fixes your problem too.

I should get the examples updated.

hecko, 2018-10-20:
What I have done is to extend the library so I can call:
`ether.client_arp_whohas_blocking(dip);
delay(10);
ether.sendUdp_mac(udp_payload, 1, nSourcePort, dip, ether.returned_mac, nDestinationPort);
`