njh/EtherCard #310: looking for a new maintainer ...

jcw, 2018-06-05:
This repository is looking for a maintainer, let me (@jcw) know if you're interested!

Jorropo, 2018-07-15:
I can do that.

jcw, 2018-07-15:
I was hoping for someone who has already contributed to this project in some way, and is known by the other contributors - i.e. @fredycpu @vicatcu @riban-bw @solarkennedy @njh - is that the case? Sorry for asking, but could you please explain what your role in the ethercard project has been so far?

njh, 2018-07-15:
I'm afraid that I have managed to accumulate too many projects to maintain already.

One option would be to configure GitHub to require X approvals from a group of reviewers before a PR can be merged, and disable pushing directly to the master branch.

jcw, 2018-07-15:
Thanks for pointing out these GitHub approval options, I didn't know about them. I would still prefer a transfer, because I have moved all my personal work to an in-house git setup (currently using "gogs"). And to be realistic about it, my contributions to this project really ended quite some years ago.

vicatcu, 2018-07-15:
I'm in a similar boat, since we stopped supporting Nanode, and moved onto wireless technologies, I don't have much in the way of personal motivation to manage the project. I'm confident someone with more "skin in the game" will step up, and thanks again for your years of stewardship over this excellent library, @jcw!

riban-bw, 2018-07-16:
I'm afraid I don't have the time to manage this. I value the existence of the library and the work done to create and maintain it but I haven't used it for a long time. I hope there can be a way that the existing codebase can remain available even if only as an archive.

njh, 2018-07-26:
While I have too many projects, I am still a big fan/user of the ENC28J60 Ethernet Controller.

So given that no other existing contributors are interested... I could take it on and review/merge PRs.

jcw, 2018-07-26:
Hello @njh - that would be grand. If you agree, I will transfer the project to you, and you can decide on things like using the approval/reviewer mechanism you suggested. It should be possible to keep it a very low-maintenance project, as it has been for me in the past several years. I'd definitely be honoured to pass on stewardship to you. -jcw

njh, 2018-07-27:
Ok, lets do it. Transfer away!

jcw, 2018-07-27:
Thank you - on its way. You'll probably want to update the opening line(s) of the top `README.md`.
Looking back, the first commit was https://github.com/jcw/ethercard/commit/e987f1e57a8173979cde956fa381718e63cb4f84 - over 8 years ago. It's great to know that it'll live on.

closed 2018-07-27 by jcw

njh, 2018-07-27:
Thank you very much for your 8 years of hard work on EtherCard.
You must have benefited so many open (and closed) source projects in that time.

I have created a `jcw` branch to mark the end of your tenure (before I ruin everything!):
https://github.com/njh/ethercard/tree/jcw