njh/EtherCard #326: Request for reviewers of Pull Requests

njh, 2018-07-30:
Hi,

I have now protected the master branch, so it is no longer possible to push directly to it. Instead Pull Requests must be raised.

The goal is to increase the code quality and ensure that automated checks are run before merging with master. I am hoping to:
- test that all the examples build automatically (this is working in Travis)
- ensure that at least one other person reviews changes before merge (reviewers can approve merges, or it would be useful for others just to give a thumbs up)
- maybe have some code style enforcement (whitespace at the end of lines etc)

The following people are currently listed at Collaborators:
* @jcw - Jean-Claude Wippler
* @solarkennedy - Kyle Anderson
* @thiseldo - Andrew Lindsay
* @vicatcu - Victor Aprea
* @Informatic - Piotr Dobrowolski
* @riban-bw - riban-bw
* @Greblys - Lukas Greblikas
* @nonflammable - Nonflammable

Nobody can push to the master branch but it does allow you to be assigned as a reviewer and maybe some other permissions.

Please let me know if you would like to help out with reviews, otherwise if I don't hear back from you, I will remove you as a collaborator (no point in too many people having access for security reasons).

There are 6 open Pull Requests, so it would be great if anyone can take a quick look and give them the thumbs up.

nick.

vicatcu, 2018-07-30:
Hey Nick, happy to stick around and try to help where I can

solarkennedy, 2018-07-30:
Please remove me

thiseldo, 2018-07-31:
I dont need to be on this as I have moved on to other areas now.
Thanks.

jcw, 2018-07-31:
I'll be happy to stick around, in the background. Nice to see some serious weeding & clean-up going on.