njh/EtherCard #330: Library manifest does not include source folder

rollercoasterdigital, 2018-08-08:
Appears to be an issue since files were moved in 8206fa1. When library is acquired by PlatformIO the source files are not included.

nonflammable, 2018-08-08:
it was reported immediately after merge:
https://github.com/njh/EtherCard/pull/327

njh, 2018-08-08:
@rollercoasterdigital which PlatformIO product do you use? PlatformIO IDE for Atom?

PlatformIO is used by Travis to build the examples - and that didn't beak when the source files were moved. But perhaps they are different products / codebases - which is a shame, because it means Travis isn't detecting faults.

rollercoasterdigital, 2018-08-08:
Yep, PlatformIO for Atom. I also tested with PlatformIO on VSCode. I think the issue is that when PlatformIO fetches the repository, it's using the `include` list in `library.json` and that doesn't include the `src` folder. The only content in the fetched library folder is `library.json` and the `examples` folder. I've made a small 
Travis is probably passing the builds when running the examples because the .h and .cpp files are in the repository and so PlatformIO doesn't perform the fetch operation each time.

njh, 2018-08-08:
I will download PlatformIO for Atom and have a go.

I have been looking through some other libraries and they don't list `include` at all.
So maybe the solution is to delete it completely.
Other are using `exclude` - maybe to remove large files?

njh, 2018-08-18:
#331 is merged now - hopefully this is fixed.

njh, 2018-09-14:
Not heard anything back, but I am going to assume that this worked and close the issue.

closed 2018-09-14 by njh