njh/EtherCard #336: Can't get IP address from DHCP

ohthoogi, 2018-08-24:
/**** enviroment ***/
 - using Aruino Uno board.
 - router :  IPTIME N702
 - example code : testDHCP.ino

/**** serial monitor  ***/
[testDHCP]
MAC: 74:69:69:2D:30:31
Setting up DHCP
DHCP failed
My IP: 0.0.0.0
Netmask: 0.0.0.0
GW IP: 0.0.0.0
DNS IP: 0.0.0.0

please give me some advice... how can i solve it?

njh, 2018-08-24:
Hm, tricky. It is difficult to know what is going on from the information you have given.

The symptoms would be the same for all of the following scenarios:

1. Ethernet cable not being plugged in
2. Your router sending DHCP packets in a slightly different format, not recognised by EtherCard
3. A fault with the ENC28J60 Ethernet controller 
4. The router being configured to ignore devices that it does't know about (MAC filtering)
5. Severe packet loss (sometimes DHCP fails with Wifi <-> EtherNet bridges on bad connections)

Maybe if you could answer some of these it might give some clues:
1.  Is the ENC28J60 board connected directly to the router?
2. Have you also tried an Arduino Ethernet Shield (W5100 / W5500 chipset)
3. Have you tried a different router?

Have you ever used a Packet sniffing tool, such as WireShark? That might give some extra information about what is going on.

ohthoogi, 2018-08-25:
Thank you for your quick reply.

/************ Reply for scenarios ******************/
1. Ethernet cable not being plugged in
  -> I plugged Ethernet cable in router and arduino LAN
2. Your router sending DHCP packets in a slightly different format, not recognised by EtherCard
  -> I will send packet please refer it. 
   I need to compare between ohter normal DHCP process and this abnormal DHCP process.
   but it's hard for me to analyze packet.. I am not good at internet protocol including DHCP.
3. A fault with the ENC28J60 Ethernet controller.
  -> I have already tested by writing mac and reading mac. 
  byte ReadMac[6];
  if (ether.begin(sizeof Ethernet::buffer, mymac, ReadMac, SS) == 0)
  of course, I tested after modifing enc28j60.* and EtherCard.*
4. The router being configured to ignore devices that it does't know about (MAC filtering)
  -> I didn't use MAC Filtering on router. and also several tested after modified mac in testDHCP.ino. 
5. Severe packet loss (sometimes DHCP fails with Wifi <-> EtherNet bridges on bad connections)
  -> I don't know for sure about that.. 
   but I don't understand why olny arduino has problem.
   so it's not very likely that sever packet loss..

/*************** Reply for clues.. *************/
1. Is the ENC28J60 board connected directly to the router?
  -> Yes, it's my connection below.
    (wi-fi)PC(USB) - (USB)Arduino(LAN) - (LAN)router(wi-fi)
       |                                                                               |  
       +----------------------------------------------------+  
2. Have you also tried an Arduino Ethernet Shield (W5100 / W5500 chipset)
  -> No, I've never used other Ethernet Shield.
3. Have you tried a different router?
  -> Yes, I did. but failed.
[Documents.zip](https://github.com/njh/EtherCard/files/2320290/Documents.zip)
[image]

njh, 2018-08-25:
The packet capture shows the Arduino successfully broadcasting a DHCP Discover packet - and as far as I can tell it looks fine.

There isn't a response from the router in the packet capture. I guess this could either be because it didn't send a response because it didn't like the Discover packet. Or because the Ethernet switch in the router filtered the packet - because the designation isn't your PC.

In the past I have used an old Ethernet *hub* - that sends all packets to all connected devices - that makes it easier to analyse what is going on.

The other thing you could try is confoguring a static IP address, and see if you are able to ping the Arduino from the PC.

ohthoogi, 2018-08-26:
Thanks a lot for giving a clue.

I tested it below in backsoon.ino

  #define STATIC 1  // set to 1 to disable DHCP (adjust myip/gwip values below)

  #if STATIC
  // ethernet interface ip address
  static byte myip[] = { 192,168,0,200 };

and then ping 192.168.0.200 from the PC.
but, there is no response...
I attached staticIP.pcapng file. please check it.
I am likely to study internet protocol from monday.
hmm.. it's difficult. 

[staticIP.zip](https://github.com/njh/EtherCard/files/2321395/staticIP.zip)

vicatcu, 2018-08-26:
I can't remember for sure, but for some reason I think the static ip octets should be in reverse order, maybe try that?

ohthoogi, 2018-08-26:
thank you for your advice.
I did it. but it didn't work. 
I found out the static ip was not in reverse order after I analyzed ARP packets in Wireshark.

njh, 2018-08-26:
It is very odd, it looks like the Arduino is able to send packets, but not receive them.

ohthoogi, 2018-08-27:
hmm..
you may be right.. It might be not a software but a hardware problem.
So I ordered the another arduino ethernet shield on the web today.
I already check communication line with oscilloscope. It look like O.K.

with below site,
http://www.handsonembedded.com/arduino-enc28j60-ethernet-module/
I tried plugging the Arduino directly into the PC
please, check it. 
arp1.pcapng : the packet after the arduino reset.
ping.pcapng : when I ping IP address from the PC.

[pcapng.zip](https://github.com/njh/EtherCard/files/2322421/pcapng.zip)
[image]

ohthoogi, 2018-08-28:
I tested it with new arduino. it works well.
but I don't know exactly why old one didn't work.
anyway.. it related to hardware problem.
thank you for supporting this question.
I attached pcapng file for a person who want to analyze DHCP Protocol in normal process.
[ArpOK2.zip](https://github.com/njh/EtherCard/files/2327688/ArpOK2.zip)

njh, 2018-08-28:
Good news! I'm glad you got it working.

closed 2018-08-28 by njh