njh/EtherCard #339: Example webClient is not working

taurean1192, 2018-09-06:
The sample code for webCliet is not working. It continuously throws the "Failed to access Ethernet Controller error"

My setup is Arduino Mega and ENC28J60 Ethernet module.

What should I do?

njh, 2018-09-06:
It sounds like it might be a problem with how you have connected your module to your Arduino Mega.

* What have you tried so far?
* Have you tried connecting to a different type of Arduino, such as an Uno?
* Do you have any details of the ENC28J60 Ethernet module?
  * The Arduino Mega is a 5v board but the ENC28J60 is a 3.3v device - do you have a level buffer?
  * The Arduino Mega 3.3 volt supply generated by the on-board FTDI chip. Maximum current draw is 50 mA.
  * The ENC28J60 maximum current draw is 180 mA

njh, 2018-09-22:
@taurean1192 are you still looking into this or should I close this issue?

njh, 2018-10-10:
No response for over 1 month, closing issue.

closed 2018-10-10 by njh

Naylamp-Mechatronics, 2019-02-25:
123