njh/EtherCard #340: EtherCard strange issue when using with LCD I2C library

UlrickDwarf, 2018-09-10:
Hi, I'm trying to develop a small program that gaher information from a web site and put info on the display, but I have some strange issues when adding the LCD libraries to the code. Here is what happen.

Starting from the WebClient example, I put a simple (local addess) in order to do some testing and the result is correct (reply HTTP 200) and I can get the results, but as soon as I add the required library in order to use the I2C LCD screen, the results is HTTP 400 bad request. Trying to investigate with tcpdump, I can see the following differences:

// the default example with no LCD header files
20:22:48.300789 IP 192.168.1.191.2850 > 192.168.1.31.80: Flags [P.], seq 0:55, ack 1, win 1024, length 55: HTTP: GET /index.html HTTP/1.0
	0x0000:  4500 005f 0000 4000 4006 b66a c0a8 01bf  E.._..@.@..j....
	0x0010:  c0a8 011f 0b22 0050 0000 0a01 7850 682d  .....".P....xPh-
	0x0020:  5018 0400 198f 0000 4745 5420 2f69 6e64  P.......GET./ind
	0x0030:  6578 2e68 746d 6c20 4854 5450 2f31 2e30  ex.html.HTTP/1.0
	0x0040:  0d0a 486f 7374 3a20 0d0a 4163 6365 7074  ..Host:...Accept
	0x0050:  3a20 7465 7874 2f68 746d 6c0d 0a0d 0a    :.text/html....

// After adding the LCD headers file
20:19:01.296292 IP 192.168.1.191.2899 > 192.168.1.31.80: Flags [P.], seq 0:73, ack 1, win 1024, length 73: HTTP: GET /index.html HTTP/1.0
	0x0000:  4500 0071 0000 4000 4006 b658 c0a8 01bf  E..q..@.@..X....
	0x0010:  c0a8 011f 0b53 0050 0000 9d01 5118 5640  .....S.P....Q.V@
	0x0020:  5018 0400 4074 0000 4745 5420 2f69 6e64  P...@t..GET./ind
	0x0030:  6578 2e68 746d 6c20 4854 5450 2f31 2e30  ex.html.HTTP/1.0
	0x0040:  0d0a 486f 7374 3a20 0408 1020 4080 0102  ..Host:.....@...
	0x0050:  0408 1020 0102 0408 1020 0d0a 4163 6365  ............Acce
	0x0060:  7074 3a20 7465 7874 2f68 746d 6c0d 0a0d  pt:.text/html...
	0x0070:  0a  

As you can see, the web call has some strange characters just after the specific request.

Here is my code:

`// Demo using DHCP and DNS to perform a web client request.
// 2011-06-08 <jc@wippler.nl>
//
// License: GPLv2

#include <Wire.h> 
#include <LiquidCrystal_I2C.h>
#include <EtherCard.h>

// ethernet interface mac address, must be unique on the LAN
static byte mymac[] = { 0x74,0x69,0x69,0x2D,0x30,0x31 };

byte Ethernet::buffer[700];
static uint32_t timer;

const char website[] PROGMEM = "192.168.1.31";

// called when the client request is complete
static void my_callback (byte status, word off, word len) {
  Serial.println(">>>");
  Ethernet::buffer[off+300] = 0;
  Serial.print((const char*) Ethernet::buffer + off);
  Serial.println("...");
}

void setup () {
  Serial.begin(115200);

  // Change 'SS' to your Slave Select pin, if you arn't using the default pin
  if (ether.begin(sizeof Ethernet::buffer, mymac, SS) == 0)
    Serial.println(F("Failed to access Ethernet controller"));
  if (!ether.dhcpSetup())
    Serial.println(F("DHCP failed"));

    ether.printIp("IP:  ", ether.myip);
//  ether.printIp("GW:  ", ether.gwip);
//  ether.printIp("DNS: ", ether.dnsip);

  Serial.println("START");
  // if website is a string containing an IP address instead of a domain name,
  // then use it directly. Note: the string can not be in PROGMEM.
  char websiteIP[] = "192.168.1.31";
  ether.parseIp(ether.hisip, websiteIP);
  //ether.persistTcpConnection(true);
}

void loop () {
  ether.packetLoop(ether.packetReceive());

  if (millis() > timer) {
    timer = millis() + 5000;
    Serial.println();
    Serial.print("<<< REQ ");
    ether.browseUrl("/", "index.html", website, my_callback);
  }
}` 

I already tried to look for similar issues, but none of them made it work.

Thanks in advance for any solution.

UlrickDwarf, 2018-09-14:
Just to update, after some more research, I finally made it work.I suppose it is related to a memory management issue as lowering the buffer it is enough to make it work.

Would it be nice to find a way to reduce memory usage by one of the library used.

closed 2018-09-14 by UlrickDwarf

njh, 2018-09-14:
Sorry for not replying sooner but my guess would be that you were running out of memory.

Glad you got it working.