njh/EtherCard #346: Arduino ProMini (328P) and ENC28J60

egntsi, 2018-11-25:
I have a small system based in a ProMini (chip Atmel MEGA328P AU1645), using a UDP server.
I tried with two 28J60 modules dated 1704 and 1415, with always the same behavior
After a IDE update to 1.8.5 (it's the same with 1.8.7), it's working only with CS = 8.
    ether.begin(sizeof Ethernet::buffer, mymac, 8);
All attempts to change it to 10 (the original CS pin) does'n work.
I have checked all IDE definitions files (pins_arduino.h) and in the Ethercard library header files and all references point to pin 10.
Anyone have any suggestions?
Any comment will be greatly appreciated.

njh, 2018-11-26:
I have successfully used EtherCard with Chip Select on Pin 8 (for Nanode boards) and Pin 10 (most other shields). In theory you should be able to use any pin for Chip Select.

Do you have anything else connected to the Pro Mini that might be using Pin 10 for something else?

egntsi, 2018-11-26:
No. The connection to CS pin of 28J60 is exclusive.
By myself I have used pin 10 in this system with no trouble. After the IDE update it only works on pin 8.
I  tried without the third parameter (default mode) an this does'n work.
I suspected there are a conflict with library definitions and something else (bootloader ??).
If I change the library default to 8, the default mode [ether.begin(sizeof Ethernet::buffer, mymac)] works.

njh, 2018-11-27:
Ah, I see, I see what you are saying now.

I am running Arduino 1.8.3 still. I will update to 1.8.7 and see if I get the same thing. I guess it is possible that Arduino have changed something to do with the way SPI works. I wonder if a git diff is possible.

While I was working on the [spi-library](https://github.com/njh/EtherCard/tree/spi-library) branch, I did notice that in once place the code is using `SS` directly, not the chosen Clip Select pin:
https://github.com/njh/EtherCard/blob/master/src/enc28j60.cpp#L235

But I don't think will cause the problem you are seeing.

I really should get that branch finished off...

egntsi, 2018-11-27:
Yes, I saw this part of the routine and tried to put pin 10 directly and also checked the SS parameter settings in the IDE routines (pins.arduino.h, ...) but the result was always the same. No way. Only works with CS = 8 !!
In short, I made the checks that were within the reach of my knowledge of the routines of the Arduino universe.
I will keep trying and will communicate any new fact. Good luck to us.
Tks Nicholas.

egntsi, 2018-11-27:
In time, on all the checks in the IDE files I always found pin 10 as default for SS

egntsi, 2018-11-29:
Well, sometimes we have surprises. The problem was bad contact on the protoboard. Honestly, I had never seen this, but china strikes again. And Murphy certainly made it happen at pin 10.
With the original application, I just put the parameters of the UDP callback function according to the new model and everything went back to normal.
before: void udpRd (word port, byte cltIP [4], const char * data, word nb);
current: void udpRd (word port, byte cltIP [4], word srcport, const char * data, word nb);
Thank you for your attention.

njh, 2018-11-30:
I'm glad you got it working.

closed 2018-11-30 by njh