njh/EtherCard #352: Not working with esp8266

chuyrg, 2019-01-10:
Hi, i've been trying to get this library working on a nodemcu board (esp8266) connected to a enc28j60, but it keeps resetting at startup.
I've read that the default cs pin makes this happen if there is a 10k pull up resistor on it, but my enc28j60 board doesn't have any.
I've also tried using a esp32 with some minor modifications to the library, but i can't get it to work on either.
Do i have to have a specific version of the esp8266 arduino core?, i'm currently using the 2.4.2 version.

njh, 2019-01-10:
Sorry but EtherCard does not currently work on non-AVR processors (ESP8266 is not AVR)

> **Requirements**
> Hardware: Only AVR-based microcontrollers are supported, such as:
>   Arduino Uno
>   Arduino Mega
>   Arduino Leonardo
>   Arduino Nano/Pro/Fio/Pro-mini/LiliPad/Duemilanove
> Any other Arduino clone using an AVR microcontroller should work
> Hardware: Non-AVR boards are NOT currently supported (101/Zero/Due)

I did do a bit of work on adding support for other architectures in Pull Request #335 - switching to using the Arduino SPI library. There might be a couple other things that need to be fixed for it to run on non-AVR.

chuyrg, 2019-01-10:
Yeah, sorry i posted this here, i thought i wa slooking into the [ethercard_ESP](https://github.com/guipoletto/ethercard_ESP) github, which is supposed to work with the esp8266.

njh, 2019-01-10:
Ok, closing this issue.

But I hope to add support for esp8266 at some point.

closed 2019-01-10 by njh