njh/EtherCard #353: Reduce the memory used by the library making printIp method optional.

mrv96, 2019-01-13:
Hi, in file webutil.cpp there is method printIp. Also if it is not used, the dynamic memory for Serial library call is used. Due to the frequent memory capacity problem on Arduino, is there a way to exclude it from the library if not used in order to reduce the dynamic memory used? I think that this issue is present only in this file and in this method. Thank you very much.

njh, 2019-01-13:
Hi,

I am not entirely clear what you are asking. 

The Arduino linker is very good at only including code that you call from your Arduino sketch. So if you don't call `printIp`, then it shouldn't pull in the `Serial` code - and reduce the Programme Memory and RAM used.

Does that help?

nick.

mrv96, 2019-01-13:
[image]
[image]

Me too believed in Arduino linker efficiency, but as you can see in the screenshots, if i comment Serial code the total amount of dynamic memory reduces. I can guarantee that i don't use printIp method in my code.