njh/EtherCard #363: httpPost ???

MatrixMonk, 2019-03-06:
static void httpPost(const char *urlbuf, 
                           const char *hoststr,
                           const char *additionalheaderline, 
                           const char *postval,
                           void (*callback)(uint8_t,uint16_t,uint16_t));

hi,
can someone point to a solution on how to pass a string to cost char* ? 
i have been away from c++ for years and have tried all the tricks that i know and nothing has worked.
tnx

MatrixMonk, 2019-03-06:
can someone explain the sense of const char* .... when one tries to send dynamic data? i am talking about const char *postval .... or is there a way to override it at run time?

xujaan, 2019-08-08:
i figured out this problem too, it is like no data was sending

GonzoAS, 2020-02-13:
example:
1 -http route of server
2- host ip 
3-  0 for default config. Content Type = xxx-form-unencoded
4 - is a string, no need to be const or PROGMEM, but it has to have html tags to work
 example->   strSecure[20] = "<html>hello bro</html>"
5- same callback functions used on examples

ether.httpPost(PSTR("/iBreader"),"192.168.54.4" ,0,strSecureB, my_callback);