njh/EtherCard #364: Does ethercard have a working limitation?

Metin81, 2019-03-10:
Dear Folks;

I have a different kind of problem. My question is that ethercard library is limited for a certain time working (such as 1hour, 24hours, etc.)...

I have a project with Ard.UNO + ENC28J60 and makes HTTP request from web and gets responses; everything is perfect for a certain time. Indeed, I have two different place (so two different networks) to check the devices let's call those places as Place_G and Place_D. (Note: The same device with the same code (sketch) used for both place.)

For Place_G
Device works perfectly only for 24 hours. (Three different times checked, and the same results came. So it is very stable in it's behavior.) I also debugging from serial monitor; after that 24 hours Atmega328 still keeps working in a good way successfully. And pings from another PCs are very successfull to ENC28J60 + Atmega328. By the way; isLinkUp function of ethercard is successfull also after 24 hours later. Only problem is that my requests from my device to web stops... I think there is something in code, which stops the sending requests after 24 hours.

The 24 hours issue so sharp, because I also measured the time in 15 seconds sensivity. Even the real time vary from the device time which came from millis() function. The "hang on" time described by millis() function in the device. Deviations with real time was 10 min. 4 min. and 2 min. respectively. So I think the stopage of sending requests steam from device itself, not from the network modem nor another component from network.

For Place_D
The same situation is valid for only 1 hour. Nothing has been changed in the device and in the code. I only moved the same structure from Place_G to Place_D. So only network changed. Stopage time moved from 24 hours to 1 hour.
Also pings were successfull... The device is replying the pings perfectly; however there is no sending requests to web again after 1 hour.

As a brief; there is something stops the perfect working of the device after a certain time (24hours for Place_G and 1 hour Place_D). All other events goes on purely with pings and with Atmega328.

After that description; Do you have any experience like I have ? What was your solution ? 

Your valuable kind answers will be appreciated....

BR 
Metin81

Metin81, 2019-03-11:
You can kindly find the code below...

#include <EtherCard.h>
const char Serial_No[] = "12345678";
static byte mymac[] = { 0x11,0x22,0x33,0x44,0x55,0x66 };
const char website[] PROGMEM = "mywebsite.com";
byte Ethernet::buffer[350];

void setup() 
{ 
  Serial.begin(57600);
  Serial.println(F("Starter Prompt"));
  Serial.println(Serial_No);

  if (ether.begin(sizeof Ethernet::buffer, mymac) == 0) Serial.println(F("Ethernet Failed..."));
  if (!ether.dhcpSetup())Serial.println(F("DHCP Failed..."));
  
  ether.printIp(F("MyIP:"),ether.myip);
  ether.printIp(F("MyGW:"),ether.gwip);
  ether.printIp(F("MyDNS:"),ether.dnsip);

  if (!ether.dnsLookup(website)) Serial.println(F("DNS Failed..."));
  ether.printIp(F("MySVR:"), ether.hisip); 
}

void loop() 
{
static uint32_t timer, Count_a, Count_b;
String Mytime;
char Data_ToSend[40];  
  
  ether.packetLoop(ether.packetReceive());

  if (millis() > timer) 
    {
    timer = millis() + 15000;
    Count_a += 1;
    if(!ether.isLinkUp()){Count_b += 1;} 
    Mytime = "CX:" + String(Count_b) + "_Tm:" + String(Count_a);
    Serial.println(Mytime);
    Mytime.toCharArray(Data_ToSend,40);
    ether.browseUrl(PSTR("/api/api.php?value="), Data_ToSend, website, callback);
    }     
}

static void callback (byte status, word off, word len) 
{ 
  Ethernet::buffer[off+len] = 0;
  Serial.println((const char*) Ethernet::buffer + off);
}

globalcitizen, 2019-03-31:
It's probably the DHCP lease expiring.

You could try:
- Use a static IP instead of DHCP (easiest fix): Recommended method if you only have one device.
- Reconfigure your DHCP server to give longer leases: Worth doing if you have many devices.
- Request a longer lease: Unsure how to do this.
- Re-request a lease after some time (eg. if over 23 hours and 55 minutes, [reset the device](https://www.instructables.com/id/two-ways-to-reset-arduino-in-software/)): Bad because it assumes a DHCP server configuration of 24 hours with no validation.

Metin81, 2019-04-01:
First of all, I thank you for your kind reply...

Most probably issue is with lease that I agree with you...
Static IP looks like a tight solution at the case. However matching with a current IP at the LAN is a problem for crowded structures...
Continuous resets are not that we want in stable workings...
Most commonly for DHCP servers lease has a limit with 1 min to 24 hours. (Some extra cases 48 hours)

antonpernisch, 2022-02-18:
Hi! First of all, I know this issue is nearly 3 years old, but this exact thing is happening to me. It's a critical system for our customer and we want to get it fixed asap. @Metin81 Do you have, please, any updates or fixes on this problem? Even a small piece of information will be useful for us... Thank you in advance :)

Metin81, 2022-02-21:
> Hi! First of all, I know this issue is nearly 3 years old, but this exact thing is happening to me. It's a critical system for our customer and we want to get it fixed asap. @Metin81 Do you have, [...]

closed 2022-02-21 by Metin81

Metin81, 2022-02-21:
After an approx. 3 years; what we have in minds. DHCP side solved by dealing a detailed digging. In case you wish we can try a compiling for you as a hex. You can send the your basic skecth to metinumar81@gmail.com and we try a hex file for you. You are welome...

globalcitizen, 2022-02-22:
One of the common issues is that if the buffer is not large enough you will see instability.

This could match the original description of issues due to differing network traffic on the two networks.

Try enlarging the buffer size.

To facilitate this without running out of SRAM try disabling features unused by editing `EtherCard.h` like:
```
#define ETHERCARD_UDP_SERVER 0
#define ETHERCARD_TCP_SERVER 0
#define ETHERCARD_ICMP 0
```

Note that setting these `#define`s in your `.ino` will not work, you must edit the source file.

To get a better idea of what takes space in your firmware you can compile it on the console with `arduino-cli` or `platformio` then use resources like:
 * `avr-nm -Crtd --size-sort  <file> |grep ' [bd] '` (shows all symbols, in descending size)
 * `avr-objdump -D -m avr <file>` (shows disassembly of all firmware regions)
 * `strings <file>` (shows all strings)
... to see what is taking space and work backwards from there.