njh/EtherCard #365: DHCP failed error

pradeepkumarbeeie, 2019-04-04:
Hi ,
I am connected the devices as mentioned above but i am getting the DHCP error like below kindly help me to resolve the same ...

[webClient]
DHCP failed
IP:  0.0.0.0
GW:  0.0.0.0
DNS: 0.0.0.0
DNS failed
SRV: 0.0.0.0

<<< REQ
<<< REQ
<<< REQ
<<< REQ
<<< REQ
<<< REQ
<<< REQ
<<< REQ
<<< REQ
<<< REQ
<<< REQ
<<< REQ
<<< REQ
<<< REQ
<<< REQ
<<< REQ

njh, 2019-04-04:
What does your setup look like? - are you connected directly to a router/DHCP server?
Which DHCP server are you using?

Would it be possible to use Wireshark to see what packets are being sent/received?

pradeepkumarbeeie, 2019-04-04:
I am Directly connected with my laptop to test the module .That time i am capturing this issue ,pls guide me resolve this ,
Whether we need to connect the module to the router only ? or any possibility to connect with the direct system for testing ?

njh, 2019-04-04:
Do you have a DHCP server running on your laptop?

davidmak2709, 2019-05-12:
Hi,

I encounter a similar problem so I would not open another issue. So I wired up everything but I really can't manage to connect the EtherCard to Internet through router or laptop. I tried the testDHCP.ino example but the only output that on the serial monitor is: 

[testDHCP]
MAC: 74:69:69:2D:30:31

Do you have any suggestion??

closed 2019-10-12 by pradeepkumarbeeie