njh/EtherCard #368: UDP not send packet

Avenitos, 2019-06-25:
Hello!

sendUdp() function not send broadcast packets :(
in Wireshark no packets.

UIP library worked correctly!

`#include <EtherCard.h>

const byte myMac[] PROGMEM = { 0x70, 0x69, 0x69, 0x2D, 0x30, 0x31 };
const char NTP_REMOTEHOST[] PROGMEM = "ntp.bit.nl";  // NTP server name
const unsigned int NTP_REMOTEPORT = 6454;             // NTP requests are to port 123
const unsigned int NTP_LOCALPORT = 8888;             // Local UDP port to use
const unsigned int NTP_PACKET_SIZE = 48;             // NTP time stamp is in the first 48 bytes of the message
byte Ethernet::buffer[350];                          // Buffer must be 350 for DHCP to work

void setup() {
  Serial.begin(57600);
  Serial.println(F("\n[EtherCard UDP send]"));

  if (ether.begin(sizeof Ethernet::buffer, myMac, SS) == 0)
    Serial.println(F("Failed to access Ethernet controller"));

  static byte myip[] = { 172,16,1,147 };
  static byte gwip[] = { 172,16,1,1 };
  static byte dnsip[] = { 8,8,8,8 };
  static byte mask[] = { 255,255,255,0 };

  ether.staticSetup(myip, gwip, dnsip , mask);

  ether.printIp("IP:  ", ether.myip);
  ether.printIp("GW:  ", ether.gwip);
  ether.printIp("DNS: ", ether.dnsip);

  char payload[] = "My UDP message";
  uint8_t nSourcePort = 6454;
  uint8_t nDestinationPort = 6454;
  uint8_t broadcast[4] = { 255,255,255,255};
  
//ether.client_arp_whohas_blocking(ipDestinationAddress);
//delay(10);
//ether.sendUdp_mac(payload, sizeof(payload), nSourcePort, ipDestinationAddress, ether.returned_mac, nDestinationPort);

  ether.sendUdp(payload, sizeof(payload), nSourcePort, broadcast, nDestinationPort);

  Serial.println("UDP sent.");  
}

void loop() {
  ether.packetLoop(ether.packetReceive());
}

`

Avenitos, 2019-06-25:
I did not immediately understand why the ntp demo works, but sending to my local server does not work. It turns out that any sending within the local segment does not work, sending via the gateway to the Internet address works normally. Why is that?

njh, 2019-06-25:
It is probably something to do with the code that chooses the Layer 2 MAC address to send to.
When sending to the Internet, EtherCard just needs the use the routers MAC address. When sending to the local subnet, it needs to find and use the right address. Although for broadcast this should just be FF:FF:FF:FF:FF:FF.

Have you tried sending to your subnet's broadcast address ? `172.16.1.255`?

See also #59

changtraidoc, 2019-06-29:
`#include <EtherCard.h>
static byte mymac[] = { 0x1A,0x2B,0x3C,0x4D,0x5E,0x6F };
byte Ethernet::buffer[700];
static uint32_t timer;
static int number=1;
const char website[] PROGMEM = "server.example.net";
const int dstPort PROGMEM = 11000;
const int srcPort PROGMEM = 4321;
void setup () {
  Serial.begin(9600);
  // Change 'SS' to your Slave Select pin, if you arn't using the default pin
  if (ether.begin(sizeof Ethernet::buffer, mymac, SS) == 0)
    Serial.println( "Failed to access Ethernet controller");
  if (!ether.dhcpSetup())
    Serial.println("DHCP failed");
  ether.printIp("IP:  ", ether.myip);
  ether.printIp("GW:  ", ether.gwip);
  ether.printIp("DNS: ", ether.dnsip);
  ether.printIp("Mask: ", ether.netmask);
//  if (!ether.dnsLookup(website))
//    Serial.println("DNS failed");
//  ether.printIp("SRV: ", ether.hisip);
    ether.parseIp(ether.hisip, "192.168.1.21");
}
// char textToSend[] = "test 123: ";
char textToSend[20];
void loop () {
    ether.packetLoop(ether.packetReceive());
    if (millis() > timer) {
      Serial.print("Numer sent: ");
      Serial.println(number);
      timer = millis() + 5000;
      Serial.println(timer);
     //static void sendUdp (char *data,uint8_t len,uint16_t sport, uint8_t *dip, uint16_t dport);

     strcpy (textToSend,"test 123: ");
     char number_t[6];
     sprintf(number_t, "%d", number);
     
     strncat (textToSend, number_t, 6);
     ether.sendUdp(textToSend, sizeof(textToSend), srcPort, ether.hisip, dstPort );

     number+=1;
     
  }
}`

marcin-koziol-iteo, 2020-06-05:
> ```c++
>  uint8_t nSourcePort = 6454;
>   uint8_t nDestinationPort = 6454;
> ```

is the same as `byte` so you're not able to set port number larger than 255