njh/EtherCard #373: Changing the default client name (Arduino-ENC28j60-BE)

hugopilot, 2019-07-09:
Hey there, quick question:
The client name is showed as 'Arduino-ENC28j60-BE'. Is there a way to change it?

hugopilot, 2019-07-09:
Sorry I didn't look good enough at `dhcpSetup`.

You can use `bool EtherCard::dhcpSetup (const char *hname)`

closed 2019-07-09 by hugopilot

Akshul-pirhoalpha, 2021-06-17:
not able to get this to work. passing the variable should be like this?

`
const char host_name[] = "github";
ether.dhcpSetup (host_name)
`

0x2b00b1e5, 2021-06-17:
You can do that. Or you could try directly parsing the string: `ether.dhcpSetup("github");` (haven't tested this, it has been quite a while since I touched C++ and this library.

Akshul-pirhoalpha, 2021-06-17:
Okay I got it. Need to pass fromRam as true. Now it's working

Akshul-pirhoalpha, 2021-07-08:
Is there a way to pass custom hostname while setting up static IP? The static IP function does not have hname as argument

Akshul-pirhoalpha, 2021-07-08:
@njh