njh/EtherCard #379: DHCP failed

Sergey477, 2019-12-27:
Hi! I'm using enc28j60 with arduino nano. (strange, but on module written "mni ecn28j60"). I am running this code:
#include <EtherCard.h>

static byte mymac[] = { 0x74, 0x69, 0x69, 0x2D, 0x30, 0x31 };
byte Ethernet::buffer[700];

void setup() {
	Serial.begin(9600);
	Serial.println("\nInitializing controller...");
	int res = ether.begin(sizeof Ethernet::buffer, mymac, 10);

	Serial.println("\nGetting DHCP info...");
	if (!ether.dhcpSetup())
		Serial.println(F("DHCP failed"));

	Serial.println("Got DHCP.");
	ether.printIp("IP:  ", ether.myip);
	ether.printIp("GW:  ", ether.gwip);
}

void loop() {
	//Serial.println(".");
}
and getting:
Initializing controller...

Getting DHCP info...
DHCP failed
Got DHCP.
IP:  0.0.0.0
GW:  0.0.0.0

I rechecked all other issues, but it didn't help.
I plugged my ecn28j60 directly to the router and my arduino nano to the computer.

njh, 2019-12-30:
What sort of ENC28J60 are you using? Is it a module that you have connected with jumpers, or a 'shield'?

Although I suspect this is most likely to be a hardware problem, have you tried using a static IP address?

Sergey477, 2019-12-30:
It is a module that I connected with wires. Yes I have tried static setup, it failed too. When I turned on debugging mode, I figured out that my router does not answer any requests. My router does not 'see' my module. I also tried UIPEthernet library, it also failed.
https://github.com/UIPEthernet/UIPEthernet

njh, 2019-12-31:
It does sound like a hardware problem.
Are you able to able to post a picture of your wiring?

oocat, 2020-03-09:
I also hit this problem, but use static ip, gw and dns is woring and not working if setup the ip only.
I using Arduino nano connected to a ENC28J60 module.

lyon470, 2020-09-27:
Hi
I have the same issue.
Is there a already a solution ?