njh/EtherCard #384: MiniCore bootloader error

mrv96, 2020-02-07:
Hi, the library works well, but when i compile with MiniCore bootloader option selected i have this error: invalid conversion from 'uint8_t* {aka unsigned char*}' to 'const char*' [-fpermissive]

The error is referred to both 'sendUdp' and 'makeUdpReply' methods. This happends because the buffer is defined as uint8_t, while the parameter 'data'of the two functions is a 'const char *'.

Since char definition sign is compiler dependent (https://stackoverflow.com/questions/29643031/invalid-conversion-from-uint8-t-aka-unsigned-char-to-const-char-fperm) and MiniCore bootloader is more recent respect both the original bootloader and Optiboot, maybe is better to update the 'data' parameter from 'const char *' to 'const uint8_t *'.

I tried this fix, and now the library compile without errors.

njh, 2020-02-09:
Thanks for letting us know.

Please could you attach a diff (or create a PR) of your change, so we can see exactly what you had to fix?

mrv96, 2020-02-09:
Done: https://github.com/njh/EtherCard/pull/385

Let me know what do you think about.