njh/EtherCard #386: Web Client Response has not enough BUFFER SPACE PLEASE HELP..

furkan-34, 2020-02-17:
Hello,
I did Web Client example. I try to get my string data from local php page with web client example. Php page only print 12345678- a couple times, but I can not see whole data. I see data length but not full data. How can I see all response string data ?

Results:
`<<< REQ >>>
HTTP/1.1 200 OK
Date: Tue, 11 Feb 2020 11:13:36 GMT
Server: Apache/2.4.41 (Win64) OpenSSL/1.1.1c PHP/7.3.11
X-Powered-By: PHP/7.3.11
Content-Length: 81
Connection: close
Content-Type: text/html; charset=UTF-8

12345678-12345678-12345678-12345678-12345678-12345678-12345678-12345678-12345678-...

njh, 2020-02-17:
The buffer size is defined by:
```
byte Ethernet::buffer[500]; // tcp/ip send and receive buffer
```

Have you tried increasing it?
What hardware are you running on?

furkan-34, 2020-02-17:
Hello,
I tried but it did not working until when I set
`bool EtherCard::persist_tcp_connection = true;` at EtherCard.cpp
` Ethernet::buffer[off+len] = 0;`
and `byte Ethernet::buffer[900]` for Arduino Nano it worked very well. 

Thank you for fast response.

closed 2020-02-17 by furkan-34