njh/EtherCard #394: enc28j60 with arduino and local web service - arduino

mahe1978, 2020-08-04:
I am trying to use Arduino Uno with ENC28j60 to send data to my local web server
the URL i used is (   http://192.168.10.6:8338/heartbeat?table=1)
when i use this URL in browser it return "OK".
with Arduino
i use this code but i can't get a result

#include <EtherCard.h>

// ethernet interface mac address, must be unique on the LAN
static byte mymac[] = { 0x74,0x69,0x69,0x2D,0x30,0x31 };

byte Ethernet::buffer[700];
static uint32_t timer;

const char website[] PROGMEM = "192.168.10.6";

// called when the client request is complete
static void my_callback (byte status, word off, word len) {
  Serial.println(">>>");
  Ethernet::buffer[off+300] = 0;
  Serial.print((const char*) Ethernet::buffer + off);
  Serial.println("...");
}

void setup () {
  Serial.begin(57600);
  Serial.println(F("\n[webClient]"));

  // Change 'SS' to your Slave Select pin, if you arn't using the default pin
  if (ether.begin(sizeof Ethernet::buffer, mymac, SS) == 0)
    Serial.println(F("Failed to access Ethernet controller"));
  if (!ether.dhcpSetup())
    Serial.println(F("DHCP failed"));

  ether.printIp("IP:  ", ether.myip);
  ether.printIp("GW:  ", ether.gwip);
  ether.printIp("DNS: ", ether.dnsip);

#if 1
  // use DNS to resolve the website's IP address
  //if (!ether.dnsLookup(website))
   // Serial.println("DNS failed");
     char websiteIP[] = "192.168.10.6";
  ether.parseIp(ether.hisip, websiteIP);

#elif 2
  // if website is a string containing an IP address instead of a domain name,
  // then use it directly. Note: the string can not be in PROGMEM.
  char websiteIP[] = "192.168.10.6";
  ether.parseIp(ether.hisip, websiteIP);
#else
  // or provide a numeric IP address instead of a string
  byte hisip[] = { 192,168,10,6 };
  ether.copyIp(ether.hisip, hisip);
#endif

  ether.printIp("SRV: ", ether.hisip);
}

void loop () {
  ether.packetLoop(ether.packetReceive());

  if (millis() > timer) {
    timer = millis() + 5000;
    Serial.println();
    Serial.print("<<< REQ ");
    ether.hisport = 8338;//to access local host
    ether.browseUrl(PSTR("/heartbeat?"), "table=3", website, my_callback);
   // ether.browseUrl(PSTR("/heartbeat?table=3"), "", website, my_callback);
    
  }
}

Serial Output :
[webClient]
IP:  192.168.10.5
GW:  192.168.10.1
DNS: 203.145.184.32
SRV: 192.168.10.6

<<< REQ 
<<< REQ 
<<< REQ 
<<< REQ 
<<< REQ 
<<< REQ 
<<< REQ 
<<< REQ

how to send " HTTP GET with request headers " Using Ethercard library(Web Client), to Local-IP Address. Thanks a lot in advance for any help/info.

nuno-silva, 2020-08-04:
I took a quick look at your code and didn't find anything wrong with it, except that you can remove the code between the `#elif 2` and `#endif` macros as well as the `website` variable.
That being said, here's a few things you can try:
- setup wireshark or tcpdump on the `192.168.10.6` machine and see if you are getting any traffic at all from the arduino
- long shot, but make sure the `mymac` MAC address is unique in your LAN (e.g. if you have any other ethercards running, make sure they have different MACs)
- try the https://github.com/njh/EtherCard/blob/master/examples/webClient/webClient.ino example without changing anything and see it that works. Then, work your way up step by step until you transform the example into what you need. **Make sure it still works with every step**. This will help finding any mistake (or bug in the library) easier along the way. E.g:
  - change from www.google.com to whatever google's IP is (e.g. `216.58.209.67`)
  - then change from google's IP to `192.168.10.6`
  - then change from port 80 to port 8338
  - then change the URL

Good luck!

mahe1978, 2020-08-05:
Hi nuno-silva
Thank you for your quick reply...
I tried with your suggestion 
I receive server reply only if domain name is "www.google.com"

ether.browseUrl(PSTR("/search?"), PSTR("source=hp&ei=I2MqX6zHLdf8rQGo7LPIAg&q=test&oq=test&gs_lcp=CgZwc3ktYWIQAzIFCAAQsQMyBQgAELEDMgIIADIFCAAQsQMyCAgAELEDEIMBMgUIABCxAzIICAAQsQMQgwEyBQgAELEDMgUIABCxAzICCAA6BQguELEDOgsILhCxAxDHARCjAjoCCC5Q2hRYshpg3x5oAHAAeACAAXaIAYADkgEDMy4xmAEAoAEBqgEHZ3dzLXdpeg&sclient=psy-ab&ved=0ahUKEwjs5IC6yIPrAhVXfisKHSj2DCkQ4dUDCAY&uact=5"), website, my_callback);

chnaging codes like
a)const char website[] PROGMEM = "192.168.10.6:8338";
 ether.browseUrl(PSTR("/heartbeat?"), "", website, my_callback);
b)const char website[] PROGMEM = "192.168.10.6";
ether.hisport = 8338;
 ether.browseUrl(PSTR("/heartbeat?"), "", website, my_callback);

and even tried
 ether.browseUrl(PSTR("/heartbeat?"), "", "http://192.168.10.6:8338", my_callback);
 ether.browseUrl(PSTR("/heartbeat?"), "", (PSTR("http://192.168.10.6:8338"), my_callback);

I can see the same old Serial Output 
[webClient]
SRV: 192.168.10.6

<<< REQ 
<<< REQ 

If anyone is able to help me, that would be great.

nuno-silva, 2020-08-05:
If you weren't able to get a response using google's IP directly, either there's an issue with `ether.parseIp` or you're doing something wrong :/ You did call `ether.parseIp`, right?

mahe1978, 2020-08-06:
Hi,
Thank you for your Suggestions.
* I tried StaticIP code example.

#include <EtherCard.h>

#define REQUEST_RATE 5000 // milliseconds

// ethernet interface mac address
static byte mymac[] = { 0x74,0x69,0x69,0x2D,0x30,0x31 };
// ethernet interface ip address
static byte myip[] = { 192,168,10,180 };
// ethernet interface ip netmask
static byte mask[] = { 255,255,255,0 };
// gateway ip address
static byte gwip[] = { 192,168,10,1 };
// remote website ip address and port
static byte hisip[] = { 192,168,10,7};
//static byte hisip[] = { 100,21,172,105}; // google port 443
//static byte hisip[] = { 3,225,178,102}; //google port 80 & 443
// remote website name
const char website[] PROGMEM = "192.168.10.7";
//const char website[] PROGMEM = "google.com";

byte Ethernet::buffer[300];   // a very small tcp/ip buffer is enough here
static long timer;

// called when the client request is complete
static void my_result_cb (byte status, word off, word len) {
  Serial.print("<<< reply ");
  Serial.print(millis() - timer);
  Serial.println(" ms");
  Serial.println((const char*) Ethernet::buffer + off);
}

void setup () {
  Serial.begin(57600);
  Serial.println("\n[getStaticIP]");

  // Change 'SS' to your Slave Select pin, if you arn't using the default pin
  if (ether.begin(sizeof Ethernet::buffer, mymac, SS) == 0)
    Serial.println( "Failed to access Ethernet controller");

  ether.staticSetup(myip, gwip, NULL, mask);
//ether.hisport = 443;//to access local host
  ether.copyIp(ether.hisip, hisip);
  ether.printIp("Server: ", ether.hisip);

  while (ether.clientWaitingGw())
    ether.packetLoop(ether.packetReceive());
  Serial.println("Gateway found");

  timer = - REQUEST_RATE; // start timing out right away
}

void loop () {
  ether.packetLoop(ether.packetReceive());

  if (millis() > timer + REQUEST_RATE) {
    timer = millis();
    Serial.println("\n>>> REQ");
   //  ether.browseUrl(PSTR("/foo/"), "bar", website, my_result_cb);
    //http://192.168.10.7
    // ether.browseUrl(PSTR("/heartbeat?"), PSTR("table=3"), website, my_result_cb);
    //ether.browseUrl("","", PSTR("http://192.168.10.7"), my_result_cb);
     ether.browseUrl("","", website, my_result_cb);
  }
}

Output:
[getStaticIP]
Server: 192.168.10.7
Gateway found

>>> REQ

>>> REQ

>>> REQ

* Using Google ip i am getting response,

Server: 3.225.178.102
Gateway found

>>> REQ
<<< reply 807 ms
HTTP/1.1 404 Not Found
Content-Type: text/plain; charset=utf-8
Date: Thu, 06 Aug 2020 10:21:15 GMT
Strict-Transport-Security: max-age=31536000; includeSubDomains; preload
Content-Length: 21
Connection: Close

default backend - 404

but i am trying with local ip there is no response.
* from server i can ping arduino static ip, from other laptop i can ping sever.
* In the laptop i tried configuration as below, there was response from server.

// ethernet interface ip address
static byte myip[] = { 192,168,10,180 };
// ethernet interface ip netmask
static byte mask[] = { 255,255,255,0 };
// gateway ip address
static byte gwip[] = { 192,168,10,1 };
// remote website ip address and port
static byte hisip[] = { 192,168,10,7};

* Library used 

	name=EtherCard
	version=1.1.0

*  I don't know why this isn't working.

nuno-silva, 2020-08-07:
If you can get a response from google's IP but not from your server's IP, possible causes are:
- a firewall is blocking your server; try accessing the web server from another host (e.g. your laptop) to see if it works;
- your server is not listening on port 80 (did you try listening on port 80 before changing the port?).

Also note that as far as I know ethercard does not support TLS/SSL/HTTP**S** (port 443).