njh/EtherCard #401: Possible incorrect logic when checking ESTAT_CLKRDY

einglis, 2020-10-30:
I was idly reading the code and spotted two similar lines:
```
byte ENC28J60::initialize (uint16_t size, const byte* macaddr, byte csPin) {
    ...
    while (!readOp(ENC28J60_READ_CTRL_REG, ESTAT) & ESTAT_CLKRDY)
        ;
```
and
```
uint8_t ENC28J60::doBIST ( byte csPin) {
    ...
    while (!readOp(ENC28J60_READ_CTRL_REG, ESTAT) & ESTAT_CLKRDY) ;
```

It's possible I've misunderstood what's going on, but shouldn't the logical AND operation be in brackets before checking false?  ie:
```
    while (!(readOp(ENC28J60_READ_CTRL_REG, ESTAT) & ESTAT_CLKRDY)) 
        ;
```

Otherwise you're just checking whether `ESTAT` is non-zero.  I suspect it works in practice, but could wrongly break out of the loop if a different bit was set in `ESTAT`.

vicatcu, 2020-10-30:
I agree... this seems like a case of missing parentheses:

```
while (!readOp(ENC28J60_READ_CTRL_REG, ESTAT) & ESTAT_CLKRDY)
```
should be...
```
while (!(readOp(ENC28J60_READ_CTRL_REG, ESTAT) & ESTAT_CLKRDY))
```
... and I honestly prefer something more like
```
while (0 == (readOp(ENC28J60_READ_CTRL_REG, ESTAT) & ESTAT_CLKRDY))
```

njh, 2020-10-31:
Thanks for reporting. I have created #402 to fix this - please can you double-check it?

I created a little test to see what the compiler thinks:

```C
#include <stdio.h>

int main()
{
   printf("!0x02 & 0x01 = ");
   if (!0x02 & 0x01) {
      printf("true\n");
   } else {
      printf("false\n");
   }
   
   return 0;
}
```

Using gcc (on Linux) - `gcc -Wall -o test.cmd test.c` - there are no errors when compiling.

But clang knows that there is a problem:

```
test.c:7:8: warning: logical not is only applied to the left hand side of this bitwise operator
      [-Wlogical-not-parentheses]
   if (!0x02 & 0x01) {
       ^     ~
test.c:7:8: note: add parentheses after the '!' to evaluate the bitwise operator first
   if (!0x02 & 0x01) {
       ^
        (          )
test.c:7:8: note: add parentheses around left hand side expression to silence this warning
   if (!0x02 & 0x01) {
       ^
       (    )
1 warning generated.
```

I don't think there is a way of using clang to compile for Arduino / AVR but it would be good if there was a way of doing static code analysis to find other bugs like this.

Anyone got any suggestions?

einglis, 2020-11-01:
No suggestions on how to coerce GCC to spot this, sorry.  The counterpoint is how impressed I am that Clang _can_ spot it!

From my perspective, I'm happy to close, but I don't know your preferred flow, so I'll just 'comment' for now.

njh, 2020-11-01:
Thanks for reporting 👍

closed 2020-11-01 by njh