njh/EtherCard #410: Example ether.browseUrl(PSTR("/foo/"), "bar", website, my_callback); not working

ColinDexter, 2021-02-06:
When I verify example webClient.ino on as eps32 with ethercard_ESP I get the following error: 

     webClient.ino: In function 'void loop()':
     
     webClient:58: error: invalid conversion from 'void (*)(byte, word, word) {aka void (*)(unsigned char, unsigned int, unsigned int)}' to 'void (*)(uint8_t, uint16_t, uint16_t) {aka void (*)(unsigned char, short unsigned int, short unsigned int)}' [-fpermissive]
     
          ether.browseUrl(PSTR("/foo/"), "bar", website, my_callback);
     
                                                                    ^
     
     In file included from webClient.ino:1:0:
     
     \Arduino\libraries\ethercard_ESP-master/EtherCard.h:411:17: note:   initializing argument 4 of 'static void           EtherCard::browseUrl(const char*, const char*, const char*, void (*)(uint8_t, uint16_t, uint16_t))'
     
          static void browseUrl (const char *urlbuf, const char *urlbuf_varpart,
     
                      ^
     
     exit status 1
     invalid conversion from 'void (*)(byte, word, word) {aka void (*)(unsigned char, unsigned int, unsigned int)}' to 'void (*)(uint8_t,      uint16_t, uint16_t) {aka void (*)(unsigned char, short unsigned int, short unsigned int)}' [-fpermissive]

njh, 2021-02-06:
I suspect that this is failing because on ESP32, the data type sizes are different to an AVR 8-bit microcontroller.

As I said in #407, I do not recommend using EtherCard with ESP devices.

ESP has a really good amount of resources and is capable of running a more complete TCP/IP stack, like LWIP.

EtherCard is intended to be used with very resource constrained (limited RAM and Program memory) Arduino devices.

ColinDexter, 2021-02-06:
Therefor I'm using the ethercard_ESP library. Which is fork is from the EtherCard library. This one is especially for the esp. Works perfectly. This can be found at: https://github.com/guipoletto/ethercard_ESP 

And there it is stated: Support The issue tracker has been moved back to Github (https://github.com/jcw/ethercard/issues) again. That's why I ask the question here.

nuno-silva, 2021-02-06:
> The issue tracker has been moved back to Github (https://github.com/jcw/ethercard/issues)

@ColinDexter that's from the original README, which was written 7 years ago, and the ethercard_ESP fork didn't update it :/
https://github.com/guipoletto/ethercard_ESP/blame/05e5b8460257b7e66df61ad0e487911e1c1d0b72/README.md#L61-L64

You should probably contact @guipoletto

njh, 2021-02-06:
I'm glad you got it working @ColinDexter. If there is a an obvious fix to the example (which isn't ESP specific), then I would be happy to accept a PR. There a probably some other changes in the `ethercard_ESP` fork that might be good to back-port.

(really need to spend some time reviewing, testing and merging the open PRs in this repo)

I am not sure why the issue tracker was turned off on the `ethercard_ESP` repo, but it is a bit confusing having issues raised here instead.

ColinDexter, 2021-02-07:
OK no problem. Thanks for the comment. Sorry for my misunderstanding. You can remove this thread if possible.

closed 2021-02-07 by ColinDexter

ColinDexter, 2021-02-15:
Hi Nicholas,

I have a problem with the DHCP. And before I make another notification on github I would like to ask you this first. I have the problem that DHCP hangs forever in the while loop on the code below.

bool EtherCard :: dhcpSetup (const char * hname, bool fromRam) {
...
     dhcpState = DHCP_STATE_INIT;
     uint16_t start = millis ();

     while (dhcpState! = DHCP_STATE_BOUND && uint16_t (millis ()) - start <60000) {
         if (isLinkUp ()) DhcpStateMachine (packetReceive ());
     }

I have a DHCP server where I do not get a DHCP lease from. And if I then unplug the network cable and then reconnect it the above code comes in a loop. The result of uint16_t (millis ()) - start gives a negative value just before it gets to the 60000 and never gets to the 60000 again. And hangs in the while loop forever.

It is of course strange that I am not getting any DHCP information. Other devices receive data from the DHCP server without any problems. But it certainly shouldn't run into a loop. And to avoid misunderstandings I have this tested with an Arduino. But with an esp it doesn't work either.

The question is if this is a known issue. And what I can do about this.

Regards Colin