njh/EtherCard #411: UDP Multicast MAC not correct?

hugopilot, 2021-02-25:
When sending something to a multicast address (e.g 225.1.2.1), a UDP packet gets properly sent (verified with Wireshark). However, my receiver doesn't get the package. 

I've tested my network and its multicasting capabilities with [this](https://www.codeproject.com/Articles/1705/IP-Multicasting-in-C) program. Using the same receiver program, it doesn't get the UDP packets from my Arduino.

I have found that the only difference between the packet sent with the example program and the Arduino is the destination MAC.

The library sets it's destination MAC to FF:FF:FF:FF:FF:FF (Broadcast). While the socket in Windows sets the destination MAC to  01:00:5E:01:02:01 (IPv4mcast_01:02:01). 

Is this why my receivers aren't recieving UDP packets from the Arduino or is it still my fault?

nuno-silva, 2021-02-28:
To receive multicast on the Arduino you need to call `enableMulticast()`:

https://github.com/njh/EtherCard/blob/39a118867afd418766d2034ca41dc83119487fe8/src/enc28j60.h#L118-L121

However, your problem seems to be *sending* from the Arduino, right?

hugopilot, 2021-02-28:
Correct

nuno-silva, 2021-02-28:
I'd say give `enableMulticast()` a try anyway. I'm not sure sending multicast packets is implemented though.

njh, 2021-03-04:
I have just done a test using [socat](http://www.dest-unreach.org/socat/) on my Mac:
```
echo "Hello World" | socat -u - udp-datagram:225.1.2.1:1234
```

And can see the packets going to `01:00:5e:01:02:01` using Wireshark.

The code that sets the MAC address for broadcast/multicast addresses is here:
https://github.com/njh/EtherCard/blob/master/src/tcpip.cpp#L384

So it indeed just sets it to FF:FF:FF:FF:FF:FF.
I am a big fan of multicast, so would like to see this working.
I am surprised that your network switch cares - what sort of switch do you have?

I have a very old Ethernet hub that I sometimes use for testing Ethercard - because it blindly copies all packets to all ports, making it easier to diagnose what is going on.

hugopilot, 2021-03-04:
> I am surprised that your network switch cares - what sort of switch do you have?

I am using a Netgear GS108. With nothing more than my PC and the arduino board connected to it. I have tried sending packets over my main network but no results so far.

I did try hardcoding the correct multicast MAC (`01:00:5e:01:02:01`) as the destination mac, but the packets didn't show up in Wireshark.