njh/EtherCard #427: Error initializing library on Arduino Nano

rbp9802, 2023-04-01:
Hello everyone, I'm trying the [backsoon](https://github.com/njh/EtherCard/blob/main/examples/backSoon/backSoon.ino) example without success. 

Program get stuck in if statement in line 46. `if (ether.begin(sizeof Ethernet::buffer, mymac, SS) == 0)` 'if' does not return any value.

From what a read from #406 and #397 I understand that this is an Arduino Nano Issue but I did not get the final solution to the problem.

I also can't upload any script to the board while my ENC28J60 module is grounded to the board (should I start another issue?). I presume it is a module wiring error but have not found any solution yet.

Hope anyone can help me, thanks!

Qix-, 2023-05-19:
Specifically, the key was the missing `SPI.beginTransaction()`. The `spi_settings` I used were:

```c++
const SPISettings spi_settings {
  250000,
  MSBFIRST,
  SPI_MODE0
};
```

barrydegraaff, 2023-06-14:
@Qix- can you send a PR or include a ZIP file here with your changes to make it work on Arduino Nano. For completeness also include your connection pinouts, specifically the CS pin is used/documented ambiguously. 

I used the stock library on Arduino Mega and confirmed it works, then I made the changes as described above, and tried on Arduino Mega, still works, verified the libraries was re-compiled. Then tried it on Arduino Nano, and it does not work.

Also I reached out to the maintainer of this repo regarding some of my earlier PRs but got no reply, so at this point I think this library is unmaintained, and I guess it is time for a fork.

Thanks very much!

Qix-, 2023-06-14:
It's definitely not enough, but that's not related to this issue. You really do need to power it with an external 3V3 supply. The low current from the Nano has nothing to do with this library hanging, though.